scientific-software-hub/waltz #84: write tabs' names in 2 rows

HelgaMond, 2018-02-22:
Make tab's name written in 2 rows. That will essentially simplify the reading of the name and free the space for some other tabs. Please, see the pic and compare two tabs with the new version and the existing one.

[image]

Ingvord, 2018-04-19:
```html
<div style="display: table-row;">
  <span style="display: table-cell;vertical-align: middle;">
<img  src= />
    </span>
<span style="display: table-cell;">
  <span style="display: block;">text1</span>
  <span style="display: block;">text2</span>
</span>
<span style="display: table-cell;">x</span>
</div>

```

Ingvord, 2018-04-21:
Closed due to #120

closed 2018-04-21 by Ingvord