scientific-software-hub/waltz #104: Feature request: Edit alias

feinfinger, 2018-03-29:
It would be great if it possible to edit aliases.

Ingvord, 2018-03-29:
@feinfinger , you can do it via database device:

1. Open sys/database/2 in Test device panel
2. Filter commands by _alias_
3. Select DbPutDeviceAlias
4. enter device and desired alias as follows: `["sys/tg_test/1","my_test"]`
5. Execute command
6. Apply device filters (to force devices tree reload) - I should probably add an implicit button for this

Ingvord, 2018-03-29:
I will think about how this can be integrated into UI

Ingvord, 2018-05-11:
Due to #123 we can add _alias_ to device info as a tree node and make it editable

Ingvord, 2018-12-14:
After discussion on Dec'13th 2018 it is decided to add ctx menu item to a device in the devices tree view: "Set alias"

closed 2019-04-03 by Ingvord

Ingvord, 2019-04-03:
@feinfinger ,

Once the new version will be deployed it will be possible to set device alias directly in device info panel:

[image]

commit 7f58e300 (2019-04-03): Progress #159 (#177)

* Progress #159 

* Close #104