scientific-software-hub/waltz #106: Plot multiple attributes in same plot

feinfinger, 2018-03-29:
Maybe this is a feature request for the more distant future: Sometimes it is very helpful to plot multiple attributes in the same window or to be able to see them at once (like measured force in x/y/z).

Ingvord, 2018-03-29:
Well, that is actually easy: replace status message (big white rectangle) with plot in monitor view; instead of opening a new tab (when plot icon is clicked) add a new trace to the plot.

feinfinger, 2018-04-03:
Hm, could you show me at some point, I don't see how to do it...

Cheers,
Fabian

Ingvord, 2018-04-03:
I mean it is not yet implemented - it is easy to implement this :)

feinfinger, 2018-04-03:
Ah, ok.

BTW: Do you prefer if I come with my ideas directly to you or is it better to post an issue on github?

Cheers,
Fabian

Ingvord, 2018-04-03:
GitHub issues must be! And we also sometimes have to discuss them. Both are totally OK, so keep going :)

commit 45319f9e (2018-04-04): Single plot for multiple scalar attributes in device monitor (#107)

Close #106
closed 2018-04-04 by Ingvord