scientific-software-hub/waltz #171: Unspecified error

feinfinger, 2018-12-19:
Sometimes unspecified errors pop up although the command is issued correctly/ This behaviour is seen when trying to move a motor.
[image]

Ingvord, 2018-12-22:
@feinfinger , thank you very much for reporting this issue!

Further info:

There is an internal server error:

```
[19/Dec/2018:09:57:27 +0100] "PUT /tango/rest/rc4/hosts/hzgpp05vme2/10000/devices/p05/tripod/pods/attributes/value?PosZ=30&async=true HTTP/1.1" 500 2589
```

Unfortunately, due to [rest-server#178](https://github.com/tango-controls/rest-server/issues/178) there is a total mess with logs (there is no logs :( ) No more information can be provided at this time

Ingvord, 2019-04-16:
Should be closed via [tango-controls/rest-server commit 07d8c26f](https://github.com/tango-controls/rest-server/commit/07d8c26f1567c61851093694d1efd879e2e109dd)

closed 2019-04-16 by Ingvord