scientific-software-hub/waltz #198: Plugins platform

Ingvord, 2019-06-04:
Waltz must support external plugins

each plufgin may :

- reside in a dedicated repo;
- has its own lifecycle
- should be easily injected into Waltz

Nodejs VS spring based apps?

Ingvord, 2019-08-30:
Technically is already achieved, see, for instance, #210 - where Waltz fetches 3rd party script from a dedicated service and then uses it. If the 3rd party script would provide webix.protoUI and use Waltz controllers that would the end result. 

To achieve that we need to extract non-UI APIs into a dedicated libraries; provide better Widget API i.e. merge webix.protoUI and Controllers logic so that client will only require to extend some class.

HelgaMond, 2019-12-10:
New architecture is planed to look like 

[image]

Ingvord, 2019-12-10:
For event bus we need to specify message format. I suggest to use dedicated issue for that - [235](https://github.com/waltz-controls/waltz/issues/235)

Ingvord, 2020-04-06:
This issue can be now closed.

Recipe for a new plugin:

1. create dedicated project - 'new-awesome-widget-war-plugin', which is either a standard war maven project OR webix jet module that produces war (see https://webix.gitbook.io/webix-jet/part-iii-practical-tasks/deploying-and-testing#deploying-on-a-back-end-server)
2. import (use npm install) dependencies like eventbus, tango-rest-client etc. 
3. develop, test etc. Use OpenAjax to subscribe to PlatformContext events etc (temporary solution)
    ```
     OpenAjax.hub.subscribe("platform_context.create", (msg, event)=>{
        //...
     });
    ```
4. package and deploy as module into Tomcat next to other plugins/waltz
5. in waltz/resources/webix_widgets/import.js add you plugin's entry point i.e. js with controller

closed 2020-04-06 by Ingvord

Ingvord, 2020-04-07:
Would be nice to have a template project for a plugin, see #242