scientific-software-hub/waltz #211: Remove Columns

feinfinger, 2019-07-30:
It would be great if it were possible to remove single columns in the table view (especially from devices that were removed from the table).
Also maybe an additional list view (as the one that was previously shown) may be helpful in some cases.

Ingvord, 2019-07-30:
@feinfinger , thank you for the issue!

Concerning removing columns in the table - there is such functionality already:

[image]

Click on the  :gear: -> click on the column name -> corresponding column will be deleted:

[image]

Ingvord, 2019-07-30:
old dashboard has been integrated into the new widget as a profile type in #215 :

[image]

closed 2019-07-31 by Ingvord