scientific-software-hub/waltz #214: Flag editable

feinfinger, 2019-07-30:
In order to avoid accidental additions to carefully predefined tables, a checkbox that makes tables editable would be helpful. That box should be unchecked (not editable) by default.

Ingvord, 2019-07-30:
@feinfinger , thanks a lot for the suggestion!

Below is how it looks like:

[image]

Settings panel is now visible by default as well. Otherwise there is a confusion - user can't do anything with a table unless he/she clicks on :gear: and set the flag. Should be fine now I guess.

Ingvord, 2019-07-31:
Should be checked by default; stored in context

commit 2e48de14 (2019-07-30): Close #214
commit 1a9266a2 (2019-08-01): Close #214
closed 2019-08-01 by Ingvord