scientific-software-hub/waltz #240: Alias not shown in Dashboard

feinfinger, 2020-04-01:
In the dashboard profiles it was possible to make a table with device attributes and the alias name of the device. This still works, but when reopening Waltz, the alias names in the dashboard are not shown anymore, but the motor tango device names are shown instead. Would it be possible to make the column Device selectable for Tango device name / alias?

Ingvord, 2020-04-01:
@feinfinger thanks for reporting this issue.

This is a known issue due to #225. The problem is that when one reopens Waltz Dashboard fetches device names from the UserContext. At that point Dashboard does not know anything about aliases... seems that easy-to-do fix is to store aliases along with device names in the UserContext.

Also adding switch between aliases and names sounds like a good idea to me. Thanks for the hint!

Ingvord, 2020-04-06:
Should be deployed with the next release!

commit 9acd3bbe (2020-04-06): Close #240
closed 2020-04-06 by Ingvord