scientific-software-hub/waltz #265: Do not support REST API 1.1

bala4901, 2021-05-22:
Hi, May I know there is support rest api 1.1

Ingvord, 2021-05-22:
@bala4901 waltz uses [tango-rest-client](https://github.com/waltz-controls/tango-rest-client) which supports only REST API v1.1

Hope this answers your question

bala4901, 2021-05-22:
I am sorry. I think I do not explain clearly. 
In document which explained Waltz support till  [https://github.com/waltz-controls/waltz#developer-requirements](RC4).

However, currently [https://github.com/tango-controls/rest-server](Tango Rest Server 2.1/2.2) is supported up to Tango REST-API version 1.1.

I had try to install using Rest-server (docker). Waltz connecteed but with json error.

[image]

I have 2 question:
1. Waltz 1.0.10 is possible to support Tango REST-API version 1.1?
2.  Where to get the detail logs for the waltz?

Best regards

Ingvord, 2021-05-22:
The document is outdated - currently supported version of Tango REST API is v1.1

> Waltz 1.0.10 is possible to support Tango REST-API version 1.1?

1.0.10 should perfectly work together with rest-server 2.1+ 

However they both must be deployed at the same servlet container i.e. put tango.war and waltz.war together into Apache Tomcat's webapps folder. 

If you want to use dockers you need to setup a proxy e.g. HAProxy like here - [see docker-compose](https://github.com/waltz-controls/waltz/blob/epics-demo/docker-compose.yml)

These are required to eliminate CORS issues 

> Where to get the detail logs for the waltz?

I use DevTools built-in to browser (usually you need to press F12)

bala4901, 2021-05-22:
thank @Ingvord I had implement as you commented. 

After implementing the HAProxy, i get some error. Can you give me some guides whether can clear the errors.
It said Data Loading error.

[image]

[image]

Ingvord, 2021-05-22:
Hm... you also need [user-context](https://github.com/waltz-controls/user-context)

I will update the docs. Thanks for reporting!

bala4901, 2021-05-22:
Hi @Ingvord 

Thank you for your support. After adding the [user-context.war](https://github.com/waltz-controls/user-context). It can be run.
I attached for my configuration for docker to run the waltz.

[waltz1_0_11-docker.zip](https://github.com/waltz-controls/waltz/files/6526912/waltz1_0_11-docker.zip)

Due to file size the WAR I do not including in the zip file:
- [waltz.war](https://github.com/waltz-controls/waltz/releases/tag/1.0.10)
- [user-context.war](https://github.com/waltz-controls/user-context)
- [tango.war](https://github.com/tango-controls/rest-server/releases/tag/rest-server-2.1)

closed 2021-05-22 by bala4901