scientific-software-hub/waltz #266: Waltz malformat for Tango rest-api 1.1

bala4901, 2021-05-22:
For those devices that not running and has errors. Waltz unable to read and will give errors.
Check in the console, it will give status 400 which is malformat.
And check on the console > Network, there is reply by Tango REST-API. but do not treat well by Waltz.
I attached the print screens for your reference

[image]

[image]

[image]

Ingvord, 2021-05-22:
@bala4901 

Thanks for the images!

This is expected behavior: for not running devices Waltz shows error notification popup (first image, top right corner). The detials are in the log (small info icon, bottom right corner)

Ingvord, 2021-06-11:
I will close this issue for now. If you have any more details/questions etc Please feel free to reopen it! Thanks

closed 2021-06-11 by Ingvord