scientific-software-hub/waltz #277: demo and docker-compose does not work

reszelaz, 2021-09-28:
Hi all,
I was going to try Waltz and I found the following problems.

1. Demo: https://github.com/waltz-controls/waltz#live-demo does not work.
2. docker-compose up fails:

```console
╰─>$ git clone https://github.com/waltz-controls/waltz.git                                                                                                                            (base) 
Cloning into 'waltz'...
remote: Enumerating objects: 10588, done.
remote: Counting objects: 100% (119/119), done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 10588 (delta 50), reused 64 (delta 13), pack-reused 10469
Receiving objects: 100% (10588/10588), 22.97 MiB | 19.41 MiB/s, done.
Resolving deltas: 100% (7214/7214), done.
╰─>$ cd waltz/                                                                                                                                                                        (base) 
╰─>$ docker-compose up                                                                                                                                                                (base) 
Creating network "waltz_public_net" with driver "bridge"
Pulling tango-db (tangocs/mysql:Database-Release-5.9)...
Database-Release-5.9: Pulling from tangocs/mysql
6ae821421a7d: Pull complete
a9e976e3aa6d: Pull complete
e3735e44a020: Pull complete
bfd564e9483f: Pull complete
df705f26e488: Pull complete
d8c3e47a6e3f: Pull complete
23dfa289d3d1: Pull complete
681bd35bda0e: Pull complete
6cf1b4c50281: Pull complete
3d50df9fe76a: Pull complete
f4773537fff9: Pull complete
244f72e330c3: Pull complete
Digest: sha256:b48180e20d39461658d1db17333260bea20e1bdeb4d259b6781cc04e06d86082
Status: Downloaded newer image for tangocs/mysql:Database-Release-5.9
Pulling tango-cs (tangocs/tango-cs:9.3.3-rc1)...
9.3.3-rc1: Pulling from tangocs/tango-cs
Digest: sha256:cc3fd9d2c5cb6d2c99dfbc01dbb84975d1ea503530f43198613ea20a38153869
Status: Downloaded newer image for tangocs/tango-cs:9.3.3-rc1
Pulling tango-test (tangocs/tango-test:9.3.3-rc1)...
9.3.3-rc1: Pulling from tangocs/tango-test
0a4690c5d889: Already exists
f35eea91b6a8: Already exists
f26d74d6d5e3: Already exists
a8f72d9a166c: Pull complete
00ace45ef2b8: Pull complete
6c3a20f6db4e: Pull complete
e0e49d5b2c5b: Pull complete
343a0700a075: Pull complete
Digest: sha256:18b2879741e112dc94e638899078d38a3f30fabb20f6051ff5b47059d3e5de44
Status: Downloaded newer image for tangocs/tango-test:9.3.3-rc1
Building waltz
Sending build context to Docker daemon     30MB
Step 1/20 : FROM node:12-alpine as BUILD
12-alpine: Pulling from library/node
6a428f9f83b0: Already exists 
d0fe2b74aff9: Pull complete 
8a7ab7725978: Pull complete 
f1c9d3375a02: Pull complete 
Digest: sha256:1ea5900145028957ec0e7b7e590ac677797fa8962ccec4e73188092f7bc14da5
Status: Downloaded newer image for node:12-alpine
 ---> b30595aabe53
Step 2/20 : ARG NODE_AUTH_TOKEN
 ---> Running in 968efc51dd55
Removing intermediate container 968efc51dd55
 ---> 9df48d256392
Step 3/20 : ARG USER_CONTEXT_VER=0.5
 ---> Running in b2c34992fc00
Removing intermediate container b2c34992fc00
 ---> 58fe8edcf0f4
Step 4/20 : ARG TANGO_REST_VER=2.2
[... 165 lines cut ...]
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, lstat '/src/node_modules/.staging/webpack-e9dfe829/schemas/plugins'
npm WARN tar ENOENT: no such file or directory, open '/src/node_modules/.staging/webpack-e9dfe829/schemas/WebpackOptions.json'
npm ERR! code E401
npm ERR! Unable to authenticate, need: Basic realm="GitHub Package Registry"

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-28T08_39_40_004Z-debug.log
The command '/bin/sh -c npm install' returned a non-zero code: 1
ERROR: Service 'waltz' failed to build : Build failed
```

Could you tell me what am I doing wrong? Thanks!

Ingvord, 2021-09-28:
@reszelaz 

> Demo: https://github.com/waltz-controls/waltz#live-demo does not work.

Yes, I shut it down as I do not want to pay for the running Amazon instance. If you want I can fire it up for a week or two.

> docker-compose up fails:

`npm install` command fails due to unauthenticated access to GitHub packages - where Waltz libraries are hosted:

> npm ERR! code E401
> npm ERR! Unable to authenticate, need: Basic realm="GitHub Package Registry"

You need to set up npm authentication, GitHub docs: [link](https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-npm-registry)

Hope this helps

reszelaz, 2021-09-28:
Thanks for the fast reply and your help!

> Yes, I shut it down as I do not want to pay for the running Amazon instance. If you want I can fire it up for a week or two.

That is not necessary but thanks for the offer.

> You need to set up npm authentication, GitHub docs: [link](https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-npm-registry)

I managed to do it. Basically one needs to generate a GitHub token with granted access to:
- repo
- read:packages

Then the said token needs to be pasted in:

https://github.com/waltz-controls/waltz/blob/d6dac0e6bca32d4e8e44e8a0e8e92fbb5b918ef9/Dockerfile#L3

Then I was trying to access to Waltz using point 2 from [this guide](https://github.com/waltz-controls/waltz#getting-started) but I could not.

I needed to obtain the Waltz container IP (using docker inspect) and navigate to it. I think the port forwarding is not enabled to the host:
```
57e6e10269a3   waltz_waltz                                                "catalina.sh run"        3 minutes ago   Up 3 minutes            8080/tcp                   waltz_waltz_1
```

I don't know if this is the correct way, but it worked for me:)

Ingvord, 2021-09-29:
@reszelaz Cool! I am glad you have managed it! And thanks for the detailed description. I will close this one for now, feel free to reopen in case

closed 2021-09-29 by Ingvord